MoonModules/MoonLight #3: P4 support

MoonModules, 2025-04-28:
Check info here

https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/get-started/index.html

ewowi, 2025-06-06:
Serial logging:

Rebooting...
ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0xc (SW_CPU_RESET),boot:0x30f (SPI_FAST_FLASH_BOOT)
Core0 Saved PC:0x4ff02280
  #0  0x4ff02280 in esp_restart_noos at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/soc/esp32p4/system_internal.c:168

Core1 Saved PC:0x4ff04bfe
  #0  0x4ff04bfe in esp_cpu_wait_for_intr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c:57 (discriminator 1)

SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x11c8
load:0x4ff2abd0,len:0xc24
load:0x4ff2cbd0,len:0x3260
entry 0x4ff2abd0
E (1028) esp-x509-crt-bundle: Invalid certificate bundle
[     2][D][Module.cpp:198] Module(): [🌙] constructor instances
[    13][D][ModuleInstances.h:38] ModuleInstances(): [🌙] constructor
[    21][D][Module.cpp:198] Module(): [🌙] constructor demo
[    27][D][ModuleDemo.h:27] ModuleDemo(): [🌙] constructor
[    50][I][esp32-hal-psram.c:102] psramAddToHeap(): PSRAM added to the heap.
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-P4
  Package           : 0
  Revision          : 1.00
  Cores             : 2
  CPU Frequency     : 360 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0000000000
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : No
  Classic BT        : No
  BT Low Energy     : No
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   584336 B ( 570.6 KB)
  Free Bytes        :   543908 B ( 531.2 KB)
  Allocated Bytes   :    33532 B (  32.7 KB)
  Minimum Free Bytes:   540104 B ( 527.4 KB)
  Largest Free Block:   385012 B ( 376.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        : 33554432 B (32768.0 KB)
  Free Bytes        : 33551856 B (32765.5 KB)
  Allocated Bytes   :        0 B (   0.0 KB)
  Minimum Free Bytes: 33551856 B (32765.5 KB)
  Largest Free Block: 33030132 B (32256.0 KB)
  Bus Mode          : QSPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
[... 57 lines cut ...]
4ff283d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff283f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff28410: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff28430: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff28450: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff28470: 0x00000000 0xbaad5678 0x00000010 0xabba1234 0x00000004 0x4ff298a8 0xbaad5678 0x0000140c
4ff28490: 0xabba1234 0x00001400 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff284b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff284d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff284f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28510: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28530: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28550: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28570: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28590: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff285b0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff285d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff285f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28610: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28630: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28650: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28670: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff28690: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 79f6151e3

troyhacks, 2025-06-06:
SD card log lines are also often a non-fatal error that just happens to be before the crash.

I'll see if I can get this booting the Moonlight skeleton on the P4 this weekend.

The "0xa5a5a5a5" is also an error pattern I saw on the M5Stack Tab 5 that's a newer ESP32-P4 device. Not entirely sure why. This is the only P4 board I have that isn't working with WLED-MM aside from serial issues on the Olimex board (which works fine with MM but doesn't log to Serial)