MoonModules/MoonLight #30: Support (DMX) lights of different type

ewowi, 2025-07-27:
MoonLight (and it's predecessors) support multiple LEDs or Lights which are all of the same type.

E.g. RGB lights, GRB lights, GRBW and since a while DMX lights like moving heads which have between 10 and 40 channels each light.
Effects and driver code is aimed at maximal efficiency and therefore a mixture of different lights is not supported yet.

In order to do so, step 1 is defining light types, where you set the number of channels for a light and assign the channels of a light to supported actions, like setRGB, setPan, setTilt etc.

step 2 is allowing different light types to be assigned to (dmx) channels

Note: idea by @troyhacks

ewowi, 2025-07-27:
Currently marked as won't as will not be implemented in or before v0.6.0 as the list of things to do for v0.6.0 is still long and September is coming closer 🙈

Merikei, 2025-11-14:
I don't need this feature, but having used [QLC+](https://www.qlcplus.org), I'm aware it has its own XML format to define DMX fixtures and has definitions for a bunch

[1] https://github.com/mcallegari/qlcplus/tree/master/resources/fixtures
[2] https://github.com/mcallegari/qlcplus-extras/tree/master/Fixtures

So mentioning it in case that helps with this